GladiusSlicer/GladiusSlicer #9: [MAINTENANCE] Do you need the f64 precision?

GodGotzi, 2024-10-08:
Some types like Transform you implemented yourself and some math functions. Is the use of f64 essential? Maybe you could use glam for more performance and more math coverage. Just a question, if I could use f32 for my usecase instead.

I read you started working on your gui again, maybe I could help here.

Almost all features a simple Slicergui could have.
uses wgpu and egui

[image]

[image]

[image]

[image]

mrhatman, 2024-10-08:
Its likely that I do not need 64 bits for precision. If I get a chance I might see if there is significant speed gains to be had with that change.

Glam is already used in my GUI so it makes sense it transition over to that long term. Its also widely used so it would make other interoperability cleaner.

FYI thank you for you interest. This code base was mostly written in isolation so a review is truly appreciated.

GodGotzi, 2024-10-20:
btw I switched to glam for everything now works fine