GladiusSlicer/GladiusSlicer #12: [BUG] Interior walls are ignored by infill

GodGotzi, 2024-10-19:
## Description

[image]

I will try fix it myself and create a pr

mrhatman, 2024-10-19:
That's interesting. 

How do you generate those interior walls? 

Does it happen with all infills?

I can take a look too if you provide a file.

GodGotzi, 2024-10-20:
It seems it happens on every infill type

When you create the walls you only subtract the exterior walls from the remaining area. 
I think this is what causes the issue.

plotter/mod.rs

```rust
fn slice_perimeters_into_chains(&mut self, number_of_perimeters: usize) {
    let mut new_chains = self
        .remaining_area
        .iter()
        .map(|poly| MultiPolygon(vec![poly.clone()]))
        .filter_map(|multi| {
            inset_polygon_recursive(
                &multi,
                &self.layer_settings,
                true,
                number_of_perimeters - 1,
            )
        })
        .collect::<Vec<_>>();

    self.fixed_chains.append(&mut new_chains);

    let perimeter_inset = if number_of_perimeters == 0 {
        0.0
    } else if number_of_perimeters == 1 {
        self.layer_settings
            .extrusion_width
            .exterior_surface_perimeter
    } else {
        self.layer_settings
            .extrusion_width
            .exterior_surface_perimeter
            + ((number_of_perimeters - 1) as f32
                * self.layer_settings.extrusion_width.exterior_inner_perimeter)
    };

    self.remaining_area = self.remaining_area.offset_from(-perimeter_inset);
}
```

mrhatman, 2024-10-20:
Thats a good point, the value .extrusion_width.exterior_surface_perimeter currently needs to match .extrusion_width.interior_surface_perimeter or there is an issue. 

plotter/infil.rs uses interior_inner_permeter
```

    let mut new_moves: Vec<MoveChain> = rotate_poly
        .offset_from(
            ((-settings.extrusion_width.interior_inner_perimeter / 2.0)
                * (1.0 - settings.infill_perimeter_overlap_percentage))
                + (settings.extrusion_width.interior_inner_perimeter / 2.0),
        )
```

The library I use for insetting only supports offsetting the entire polygon where this would require insetting them independently. 

The implementation should be made to only use the exterior value for both and a warning or error should be created when they do not match.