GladiusSlicer/GladiusSlicer #26: [MAINTENANCE] Document the algorium

mrhatman, 2024-11-28:
## Description

The documentation on the main triangle tower in non-existent. This needs to be fixed as the project expands.

## Reason

The code should be understandable by all and its complexity should be understood.

mrhatman, 2024-11-28:
The rough steps of the algorithm. Will expand each step in future posts

The basic idea is to start at the bottom of the model and progress upwards maintaining a list (rings) of the currently intersected edges and triangles. As you move upwards, on each vertex remove any triangle or edge that ends with that point and add all fragments that start at that point. Then combine with the fragments from the original ring to make the new ring(s). At any point you can get the slice by iterating around the rings. 

1. For each vertex find the ring fragments that begins at that vertex.
2.  Place vertices in priority queue
3. Starting at the lowest vertex, remove points until you reach slice height
4. Whenever you remove a point, split the active ring on any edges that end with that point and any triangles that lie between to edges that end on that point. Then add the new fragments for that point to the split ring fragments. Combine these fragments by connecting matching edges and triangles to each other. this should form the new active ring.

Diagrams and more detailed steps to come.

mrhatman, 2024-12-04:
The rough complexity of this is ( V+L) S + VlogV

Where V is vertex count, L is layer count and S is average number of triangles in a layer.

chungwong, 2024-12-09:
While we are on the topic of documenting the slicing algorithm. Do you have any reference for the support and infill algorithm?
I can see that support is genereated in [`add_support_polygons`](https://github.com/GladiusSlicer/GladiusSlicer/blob/main/src/plotter/support.rs#L13) while infill is genereated at https://github.com/GladiusSlicer/GladiusSlicer/blob/main/src/plotter/infill.rs#L126

mrhatman, 2024-12-09:
Support is currently not a great algorithm 
Basically it just iterates the slices from top to bottom. If any area would not be supported it adds interface material, if an area needs support for the iterface material or a support tower, add a support tower. i would love to use a better algorithm for the base towers and to support an algorithm like support trees.

The infill is calculated by splitting the polygon into  [y-monotone regions](https://en.wikipedia.org/wiki/Monotone_polygon) These allow that the plotter can just go back and forth with no fear of intersections. The polygon can be rotated and spacing changed to allow for all of the infill provided. Note that not linear infill like hexagon and gyroid are not supported yet.

chungwong, 2024-12-10:
> i would love to use a better algorithm for the base towers and to support an algorithm like support trees.

It doesn't look like Cura documented the tree support in anyway(I cannot find it). It is difficult to find exsting documentation on support algorithm. Fortunately I did found a review on this subject, https://www.mdpi.com/2504-4494/2/4/64, page 10, Table 2

mrhatman, 2024-12-10:
> It doesn't look like Cura documented the tree support in anyway(I cannot find it). It is difficult to find exsting documentation on support algorithm. Fortunately I did found a review on this subject, https://www.mdpi.com/2504-4494/2/4/64, page 10, Table 2

It appears that while Cura does not document the algorithm, the original fork does have some documentation that seems ok. https://github.com/ThomasRahm/CuraEngine#how-the-algorithm-generating-the-trees-works.